nanopb/nanopb #1073: Size not generated for field with both max_size and FT_CALLBACK

geoffreymbrown, 2025-03-16:
I've been using the generated message size fields as a compiler check for static buffer allocation.   I'm curious why the following does not have a generated size field.  In other words, why can't the max_size be interpreted as a "contract" between the application and nanopb. 

message PT_RAW {
    fixed32 epoch = 1;
    fixed32 voltage = 2;
    fixed32 temperature = 3;
    bytes samples = 4;
  }

PT_RAW.samples max_size:240
PT_RAW.samples type:FT_CALLBACK

PetteriAimonen, 2025-03-16:
Well, it just hasn't been designed to be. Currently the maximum size is only defined when nanopb is in control of the message size.

This may be related to what was being asked in #1059, so there may be some point for a change in behavior.

geoffreymbrown, 2025-03-16:
Thank you for getting back.  I'd like to respectfully argue for the change.   I'm guessing that in many cases a protobuf message to be sent will be encoded before sending.  Without having a bound on the size of the message, it's hard to safely allocate the needed transmit buffer.   It's true, that the implementation of the callback should respect any memory constraints, but without the maximum size,  a program has less information to respect those constraints.

PetteriAimonen, 2025-03-16:
Well, as long as the callback uses `pb_write()`, it will properly return an error if the allocated buffer gets full.

But it seems there is demand for this kind of feature. I'm not sure of a reasonable implementation that wouldn't cause confusion (it should be made clear when the maximum encoded size is truly the maximum, and when does it depend on user code).