nanopb/nanopb #1083: Encoding nested oneof message takes one extra byte than reported by the max size

iniw, 2025-04-22:
Given the following protobuf definitions:
```proto
message HeatingReport {
  float celsius = 1;
  float seconds_elapsed = 2;
  bool finished = 3;
}

message Alarm {
  int32 error_code = 1;
}

message FirmwareEvent {
  oneof tag {
    HeatingReport heating_report = 1;
    Alarm alarm = 2;
  }
}
```
the nanopb generator reports that the "Maximum encoded size of messages (where known)" is 13 bytes:
```cpp
/* Maximum encoded size of messages (where known) */
#define FirmwareEvent_size                       13
#define PROTO_FIRMWARE_FIRMWAREEVENT_PB_H_MAX_SIZE FirmwareEvent_size
```
But the following code fails to encode, complaining about not having enough space in the stream (generated with `anonymous_oneof:true`, hence the terser initialization):
```cpp
FirmwareEvent event {
    .which_tag = FirmwareEvent_heating_report_tag,
    .heating_report = {
        .celsius = 1.0f,
        .seconds_elapsed = 1.0f,
        .finished = true,
    },
};

uint8_t buffer[FirmwareEvent_size];

auto stream = pb_ostream_from_buffer(buffer, sizeof(buffer));
if (!pb_encode(&stream, FirmwareEvent_fields, &event)) {
    size_t encoded_size;
    assert(pb_get_encoded_size(&encoded_size, FirmwareEvent_fields, &event));

    LOGE("Bridge", "Encoding failed (error=\"{}\", bytes_available={}, bytes_needed={})", PB_GET_ERROR(&stream), sizeof(buffer), encoded_size);
    return;
}                                                                                                                                            
```
And the resulting log is:
```
Encoding failed (error="stream full", bytes_available=13, bytes_needed=14)
```
My question is: why does that need 14 bytes? `sizeof(HeatingReport)` reports 12, why does the tag need 2 bytes? And even then, why is the reported max size 13 instead of 14? Is there some alignment requirement that I'm not aware of?

PetteriAimonen, 2025-04-23:
The C `sizeof()` doesn't have much relation to the encoded protobuf size, the encodings are so different.

The correct maximum size for the message is 14 bytes:
* 1 byte for the tag of `FirmwareEvent.heating_report`
* 1 byte for the length of `HeatingReport` submessage
* 1 + 4 bytes for each of the two floats in `HeatingReport`
* 1 + 1 bytes for the bool in `HeatingReport`

Which version of nanopb are you using? I get `#define FirmwareEvent_size    14` on latest git master, on 0.4.9 and on 0.3.9.

iniw, 2025-04-23:
In my testing I nuked my build folder and it now gives me 14 on both versions. I think know what happened here.

In the real code, all 3 messages in the OP are defined in different files. When I added the `finished` field to `HeatingReport`, nanopb re-generated `HeatingReport.pb.h` but _not_ `FirmwareEvent.pb.h` since it's contents didn't change, even though the maximum size of `FirmwareEvent` depends on the maximum size of `HeatingReport`.

In my testing I nuked the build folder a couple times to test the different nanopb versions, which forced the regeneration of all files and fixed the issue, but the problem wasn't in the nanopb version, it was in the cached generator output.

I don't think nanopb generates a graph of dependencies, since that would require parsing `import` statements in the protobuf definitions. The "fix" is probably to just delete and re-generate all of the files in every build.

If this was a c++ library you could mark `pb_get_encoded_size` as `constexpr` and define `{MessageName}_size` as the sum of the encoded size of all it's fields instead of a generation-time hardcoded value, but I digress :)

iniw, 2025-04-23:
Did some further testing and concluded that this in fact the issue. Changing `HeatingReport.proto` causes `HeatingReport.pb.h` to regenerate but not `FirmwareEvent.pb.h`.

> The "fix" is probably to just delete and re-generate all of the files in every build.

I attempted to look into this but didn't really find an easy way to do it. I think I'm just gonna have to be mindful of changes to the proto files, dunno.

iniw, 2025-04-23:
> I don't think nanopb generator even does any such caching, that is probably part of whatever build system rules you are using. The build system should be set up to track dependencies.

I just followed the CMake example:

```cmake
set(NANOPB_OPTIONS "-s anonymous_oneof:true")

find_package(Nanopb REQUIRED)

file(GLOB_RECURSE PROTO_FILES ../../shared/proto/*.proto)

nanopb_generate_cpp(
    TARGET proto
    RELPATH ../../shared
    ${PROTO_FILES}
)

target_link_libraries(project PRIVATE proto)
```

I find it hard to believe that CMake has the information that `FirmwareEvent.pb.h` depends on `HeatingReport.pb.h`. Looking at the code on `FindNanopb.cmake` each generated C file depends only on the `.proto/.optoin` file it's generated from (and the generator files themselves).

https://github.com/nanopb/nanopb/blob/e00d04e2c575c19b7e582c8713be2a42098a8c87/extra/FindNanopb.cmake#L313-L326

PetteriAimonen, 2025-04-23:
I think the `NANOPB_DEPENDS` option could be used for this.

But I don't have a strong opinion on this, I don't use CMake much.

iniw, 2025-04-23:
> I think the `NANOPB_DEPENDS` option could be used for this.

Huh, somehow I completely missed this variable when looking through the CMake code. Setting this to the list of proto files does essentially what I want. Thanks a lot.

closed 2025-04-23 by iniw