nanopb/nanopb #1085: Input streams with unknown length always result in failure

MrSurly, 2025-04-26:
Per the [documentation](https://jpa.kapsi.fi/nanopb/docs/concepts.html#input-streams):

> You don’t need to know the length of the message in advance. After getting EOF error when reading, set bytes_left to 0 and return false. pb_decode() will detect this and if the EOF was in a proper position, it will return true.

Scenario 1: Initially set `bytes_left` to `SIZE_MAX` , and return `false`/`0` when EOF is reached: `io error`
Scenario 2: Initially set `bytes_left` to `36` (actual message size) and return `false`/`0` when EOF is reached: `io error`
Scenario 3: Initially set `bytes_left` to `SIZE_MAX` and always return `true` w/o changing `bytes_left`: `zero tag`
Scenario 4: Initially set `bytes_left` to `36` (actual message size) and always return `true` w/o changing `bytes_left`: Success

I briefly looked at the (0.4.9.1) code, and only found 2 places where `stream.callback` was being called, and both looked like this:

    if (!stream->callback(stream, buf, 1))
        PB_RETURN_ERROR(stream, "io error")

It appears that it will _always_ issue an `io error` if the callback returns `false`, which agrees with what I'm seeing.

My test code: [nanopb_test.tar.gz](https://github.com/user-attachments/files/19924902/nanopb_test.tar.gz)

For the sake of file size, the tarball above does not include the nanopb directory found here: https://jpa.kapsi.fi/nanopb/download/nanopb-0.4.9.1-linux-x86.tar.gz

PetteriAimonen, 2025-04-27:
I didn't try your test case yet, but the functionality is covered by this test case:
https://github.com/nanopb/nanopb/blob/master/tests/basic_stream/decode_stream.c

The functionality is implemented here (https://github.com/nanopb/nanopb/blob/master/pb_decode.c#L278-L316) which I agree is a bit of a hack due to changes made over the years, but should work as documented.

PetteriAimonen, 2025-04-27:
I think the issue with your code may be that it returns `false` after the last write of exactly the number of bytes left in the buffer actually succeeds.

I would also recommend returning `false` if `read_cb()` is called after the buffer is already empty or contains less than `count` bytes. Such a read can happen if the message is corrupted, and I think your current code would end up reading past your buffer.

MrSurly, 2025-04-27:
This worked, which returns `false` _after_ the npb code sets `bytes_left` to (I think) -37:

```
bool read_cb(pb_istream_t *stream, uint8_t *buf, size_t count) {
    static uint8_t *cur = data;
    bool ret = true;
    
    printf("count: %ld data_cnt: %d\n", count, data_cnt);
    if (data_cnt < 1) {
      printf("EOF, return false\n");
      stream->bytes_left = 0;
      return false;
    }
    while(count--) {
        *buf++ = *cur++;
        data_cnt--;
    }

    return true;
}
```

```
36 bytes written
bytes left: 18446744073709551615
count: 1 data_cnt: 36 left: 18446744073709551615
count: 1 data_cnt: 35 left: 18446744073709551614
count: 1 data_cnt: 34 left: 34
count: 1 data_cnt: 33 left: 33
count: 1 data_cnt: 32 left: 32
count: 1 data_cnt: 31 left: 31
count: 1 data_cnt: 30 left: 30
count: 1 data_cnt: 29 left: 29
count: 1 data_cnt: 28 left: 28
count: 1 data_cnt: 27 left: 27
count: 1 data_cnt: 26 left: 26
count: 1 data_cnt: 25 left: 25
count: 1 data_cnt: 24 left: 24
count: 1 data_cnt: 23 left: 23
count: 1 data_cnt: 22 left: 22
count: 1 data_cnt: 21 left: 21
count: 1 data_cnt: 20 left: 20
count: 1 data_cnt: 19 left: 19
count: 1 data_cnt: 18 left: 18
count: 1 data_cnt: 17 left: 17
count: 1 data_cnt: 16 left: 16
count: 1 data_cnt: 15 left: 15
count: 1 data_cnt: 14 left: 14
count: 1 data_cnt: 13 left: 13
count: 1 data_cnt: 12 left: 12
count: 1 data_cnt: 11 left: 11
count: 1 data_cnt: 10 left: 10
count: 1 data_cnt: 9 left: 9
count: 1 data_cnt: 8 left: 8
count: 1 data_cnt: 7 left: 7
count: 1 data_cnt: 6 left: 6
count: 1 data_cnt: 5 left: 5
count: 1 data_cnt: 4 left: 4
count: 1 data_cnt: 3 left: 3
count: 1 data_cnt: 2 left: 2
count: 1 data_cnt: 1 left: 1
count: 1 data_cnt: 0 left: 18446744073709551579
EOF, return false
read status: OK
bytes left: 0
which_regs: 2
cell cnt: 15 1st cell: 17
```

MrSurly, 2025-04-27:
Okay, now I know it's the "read past end to get EOF" idiom, this is working.

closed 2025-04-27 by MrSurly

PetteriAimonen, 2025-04-28:
No problem. You are free to set `bytes_left` to 0 already on the last successful read, if you know that it is the end, but then return `true` to indicate the last read was successful.