nanopb/nanopb #1086: Bug: `FT_CALLBACK` for single nested message not working inline in `.proto` file

rbbag, 2025-04-29:
I have encountered an issue where using `(nanopb).type = FT_CALLBACK` for a single nested message directly in the `.proto` file does not produce the expected `pb_callback_t` member in the generated C structure. However, defining the same using an `.options` file works correctly. This suggests a potential bug in the nanopb generator.

Here is a simplified version of my `.proto` definition:

```protobuf
syntax = "proto3";

import "nanopb.proto";

message InnerTestMessage {
  uint32 value = 1;
}

message TestMessage {
  uint32 request_id = 1;
  InnerTestMessage inner_message = 2 [(nanopb).type = FT_CALLBACK];
}
```

With nanopb version `0.4.8-dev`, the generated C header (`.pb.h`) for `TestMessage` incorrectly looks like this:
```c
typedef struct _TestMessage {
  uint32_t request_id;
  bool has_inner_message;
  InnerTestMessage inner_message; /* Message */
} TestMessage;
```
The desired output, which is correctly generated when using an `.options` file, is:
```c
typedef struct _TestMessage {
  uint32_t request_id;
  pb_callback_t inner_message; /* Message */
} TestMessage;
```

## Steps to reproduce:
1. Create the `.proto` file as shown above.
2. Compile it using nanopb version `0.4.8-dev` with the command:
```bash
python nanopb_generator.py --output-dir=%OUTPUT_DIR% %PROTO_FILE% -e ".npb" -s long_names:true
```
3. Observe the generated `.pb.h` file. The `inner_message` member is not `pb_callback_t`.
4. Create an `.options` file with the content:
```
TestMessage.inner_message type:FT_CALLBACK
```
5. Recompile the `.proto` file (nanopb should automatically pick up the `.options` file).
6. Observe the generated `.pb.h` file. The `inner_message` member is now correctly `pb_callback_t`.

This inconsistency indicates that the (nanopb).type = FT_CALLBACK directive for a single nested message is not being correctly processed when defined inline in the .proto file.

I hope this report helps in identifying and fixing this issue.

PetteriAimonen, 2025-04-29:
I can't seem to be able to reproduce this on 0.4.8 nor 0.4.9.

You can try passing `--verbose` to `nanopb_generator` to get more debugging output.

<pre>
petteri@raichu:~/nanopb-test/nanopb-0.4.9-linux-x86/test_1086$ ls
test_1086.proto
</pre>

<pre>
petteri@raichu:~/nanopb-test/nanopb-0.4.9-linux-x86/test_1086$ cat test_1086.proto 
syntax = "proto3";

import "nanopb.proto";

message InnerTestMessage {
  uint32 value = 1;
}

message TestMessage {
  uint32 request_id = 1;
  InnerTestMessage inner_message = 2 [(nanopb).type = FT_CALLBACK];
}
</pre>
<pre>
petteri@raichu:~/nanopb-test/nanopb-0.4.9-linux-x86/test_1086$ python ../generator/nanopb_generator.py -e '.npb' -s long_names:true test_1086.proto 
Writing to test_1086.npb.h and test_1086.npb.c
</pre>
<pre>
petteri@raichu:~/nanopb-test/nanopb-0.4.9-linux-x86/test_1086$ head -n 30 test_1086.npb.h 
/* Automatically generated nanopb header */
/* Generated by nanopb-0.4.9 */

#ifndef PB_TEST_1086_NPB_H_INCLUDED
#define PB_TEST_1086_NPB_H_INCLUDED
#include <pb.h>

#if PB_PROTO_HEADER_VERSION != 40
#error Regenerate this file with the current version of nanopb generator.
#endif

/* Struct definitions */
typedef struct _InnerTestMessage {
    uint32_t value;
} InnerTestMessage;

typedef struct _TestMessage {
    uint32_t request_id;
    pb_callback_t inner_message;
} TestMessage;

#ifdef __cplusplus
extern "C" {
#endif

/* Initializer values for message structs */
#define InnerTestMessage_init_default            {0}
#define TestMessage_init_default                 {0, {{NULL}, NULL}}
#define InnerTestMessage_init_zero               {0}
</pre>

rbbag, 2025-04-30:
Thanks for the response.

I've found that the issue was due to using the development version 0.4.8-dev. After updating to the stable version 0.4.9, everything works as expected. It seems there was some specific behavior in the development version that caused this problem.

Apologies for the unnecessary report; the issue was on my end. You can close this issue.

rbbag, 2025-04-30:
Can I ask you one more question? Is it possible to somehow set all submessages to have FT_CALLBACK type?

Thanks

PetteriAimonen, 2025-05-01:
You can specify options at message or file level, and they will apply to all fields. It's also possible to e.g. specify `FT_CALLBACK` on message level and then override it to `FT_STATIC` for the fields you don't want to be callbacks:

https://github.com/nanopb/nanopb/blob/master/tests/options/options.proto

closed 2025-05-01 by PetteriAimonen