nanopb/nanopb #1092: No package.json in nanopb GitHub leads to MissingPackageManifestError: Could not find one of 'package.json' manifest files in the package

Scandroid-Monochrome, 2025-05-18:
Hi Petteri,

I had a MissingPackageManifestError for my nanopb implementation in PlatformIO, and I know that this error is caused by not having the correct .json file to interface PlatformIO and the library involved. In this case, package.json was missing. 

I looked in your GitHub for the package.json file, and there was none when I typed it into the search bar(I tried making sure that items in folders came up when typing in the search bar). 

Is there any fix to this or is package.json just missing from the GitHub? Either I need package.json for nanopb or some way around having to use package.json.

Thank you for your time,
- Scandroid-Monochrome

PetteriAimonen, 2025-05-19:
https://github.com/nanopb/nanopb/tree/master/examples/platformio builds ok for me.

Scandroid-Monochrome, 2025-05-19:
So are we supposed to download from the platformio folder instead of the nanopb folder as the root directory?

PetteriAimonen, 2025-05-19:
No, but you can take a look at that example to see how to configure your 
platformio.ini.