nanopb/nanopb #1096: why user encode callback can't change the substream.bytes

JackChan-WO, 2025-06-05:
Hi nanopb,
In the pb_encode.c #753 
```
    size = substream.bytes_written;
    /* Use a substream to verify that a callback doesn't write more than
     * what it did the first time. */
    substream.callback = stream->callback;
    substream.state = stream->state;
    substream.max_size = size;
    substream.bytes_written = 0;
#ifndef PB_NO_ERRMSG
    substream.errmsg = NULL;
#endif
    
    status = pb_encode(&substream, fields, src_struct);
    
    stream->bytes_written += substream.bytes_written;
    stream->state = substream.state;
#ifndef PB_NO_ERRMSG
    stream->errmsg = substream.errmsg;
#endif
    
    if (substream.bytes_written != size)
        PB_RETURN_ERROR(stream, "submsg size changed");
```
the substream will be encoded by the user defined callback function and the the bytes_written will be changed, but in code here it looks like it checks the value before and after the callback function and reports error if the value is changed. I am wandering why and can you provide a sample to encode the submessage without changing the bytes_written vaule

PetteriAimonen, 2025-06-05:
In protobuf format, the length of a submessage must be encoded before the contents of the submessage.
That's why the callback needs to be called twice, as described [in the documentation](https://jpa.kapsi.fi/nanopb/docs/concepts.html#encoding-callbacks).

The check makes sure that when the callback is called for second time, it writes out the same number of bytes as it did on the first call.

PetteriAimonen, 2025-06-06:
I couldn't understand your description.

Maybe this example would shed some light?
https://github.com/nanopb/nanopb/blob/master/examples/network_server/server.c#L26-L62

JackChan-WO, 2025-06-06:
@PetteriAimonen  Thanks, I think I am fine for this issue. We can close it.