nanopb/nanopb #1099: rearranging structures in pb.h

pavithramaiya, 2025-06-18:
Hi, I'm using 0.4.9.1 version.

input in .proto
message temp
{
    optional uint32     xyz1= 1; 
    optional uint32     xyz2= 2;  
}

generated pb.h:
typedef struct _bdd_temp{
    bool has_xyz1l;
    uint32_t xyz1;
    bool has_xyz2;
    uint32_t xyz2;
} bdd_temp;

but to optimize the memory, i want to change it to 
typedef struct _bdd_temp{
uint32_t xyz1;
 uint32_t xyz2;

    bool has_xyz1l;
    bool has_xyz2;
   
} bdd_temp;

any suggestions how to change is automatically or manually and any impact on other files?
currently i get error FIELDINFO_DOES_NOT_FIT_width1_field2

pavithramaiya, 2025-06-19:
It did not. This used to work in 0.3.2 without any discriptorsize.

PetteriAimonen, 2025-06-19:
Yeah, there is no test coverage or official support for this kind of rearrangement.